atim-radiocommunications/armapi #2: Activation OTAA défectueuse ?

Mobiak, 2016-04-21:
Y-a-t-il des fonctionnalités LoRaWAN qui ne sont pas supportées par le module N8-LW ?

J'essaye d'activer le mode OTAA, mais lorsque j'active le mode avec

myArm.LwEnableOtaa(true);
myArm.UpdateConfig();

Lorsque j'interroge le module ensuite avec myArm.LwIsEnableOtaa(), la valeur retournée reste à 0.

En mode personnalisé, la transmission de données se passe correctement.

Le module est :
Rev:15.1124.1700 S/N:XXXXXXXXXXXXXXXX LoRaWan:V1.0 868MHz:14dBm

franckCordu, 2017-08-23:
Bonjour,

Avez vous trouvé une solution, j'ai moi aussi ce problème. j'aimerai utiliser l'OTAA sur un N8_LW et je n'y parvient pas.

Merci,

Franck

ClementD74, 2021-07-05:
@franckCordu see my answer here : https://github.com/atim-radiocommunications/armapi/issues/14
It should work.