atim-radiocommunications/armapi #3: ARMAPI on Arduino Micro

alexissusset, 2016-06-29:
When working with the armapi library on Arduino Micro, it seems that everything is working fine using the following init command (using the Uplink example and changing Serial to Serial1):
if (myArm.Init(&Serial1) != ARM_ERR_NONE)

But later when using the send API, Sigfox doesn't seem to receive any message:
myArm.Send(msg, sizeof(msg));

Is there a way to get the status of the send command?

antoine163, 2016-06-30:
Can you check value retrouned by [SetMode](
http://atim-radiocommunications.github.io/armapi/doc/html/a00026.html#ga68d7f197f155e378c17b78c29cc3ad4f) and [UpdateConfig](http://atim-radiocommunications.github.io/armapi/doc/html/a00026.html#ga9e56b8b4a573f76f7ceb1776b5ffec33) method ?

alexissusset, 2016-06-30:
Hi @antoine163 

Here are the results for SetMode and UpdateConfig:
Set Mode:
0
Update Config Mode:
0

I used the following code:
  Serial.println("Set Mode:");
  Serial.println(myArm.SetMode(ARM_MODE_SFX));
  myArm.SfxEnableDownlink(false);
  Serial.println("Update Config Mode:");
  Serial.println(myArm.UpdateConfig());

antoine163, 2016-06-30:
Ok, I think the library work correctly.
What is the shield you use ?
When you tray the "sigfox_uplink.ino" example, the led blink ?
You receive '13' on backend ?

alexissusset, 2016-06-30:
I've connected the (Sigbee) module via a Waveshare bee adaptor (http://www.waveshare.com/xbee-usb-adapter.htm)
It is connected to Arduino Micro Serial RX/TX as well as +3.3V and Ground

When I try the sigfox_uplink.ino, the LED on Arduino stays off
On Sigfox backend side, I am not getting anything, my device appears as last seen: "N/A"

As a side note, a LoRaWAN chipset works perfectly with the same pin connections

antoine163, 2016-07-01:
> When I try the sigfox_uplink.ino, the LED on Arduino stays off

Ok, and on the module ?
Can you check you Sigfox ID ? Compare your ID and the ID in the module with [armInfo](http://atim-radiocommunications.github.io/armapi/doc/html/a00026.html#ga1fa4df13601675bb49713201d9336923) method.

--------------------------------------------------------------------
```
uint64_t id=0;
myArm.Info(0, 0, &id, 0, 0);
Serial.print(id, HEX);
```

alexissusset, 2016-07-01:
Yes the ID matches the ID printed on the module and inside Sigfox Backend

antoine163, 2016-07-01:
The led on module blink? (with sigfox_uplink.ino example)
You been covered by Sigfox network?

alexissusset, 2016-07-01:
Yes, the LED on the module blinks.
And I am certain Sigfox works in the area as there is a working Sens'it in the same room as the ATIM Sigbee

closed 2016-07-28 by antoine163