atim-radiocommunications/armapi #4: sigfox downlink message not working

joaqrus, 2016-10-23:
I have recently updated the firmware of my ARM N8 LP module to the last version, and I can send correctly uplink/downlink messages to the Sigfox backend using standard AT commands, such as:

**Uplink message:**
`AT$SF=112233445566778899AABBCC,0`

**Downlink message:**
`AT$SF=112233445566778899AABBCC,1`

However when using the armapi arduino libraries I am having two different issues. I am using an Arduino Zero board that has an Atmel ARM Cortex M0+ microcontroller, where the serial connection is called Serial1, and the native USB connection SerialUSB.

My Arduino sketch works using uplink messages, but I have two issues:

1. When I try to send a message in the setup code section, to pull the date/time from the sigfox backend, this first downlink message does not work. I think it could have something to do with the fact that there is a 3 seconds boot time in the ARM N8 module.

2. The downlink message never works using the Arduino library. It is always received as an uplink message, even though I configure the module to send a Downlink message.

This code is coming from your original sigfox downlink example:
```
#include <arm.h>
#define LED 13

//Instance of  the class Arm
Arm myArm;
//The message to send at sigfox
uint8_t msg[] = "Hello Sigfox";
//Buffer for data sigfox downlink
uint8_t data[8] = {0};

void setup()
{
  SerialUSB.begin(115200);              // Native USB 
  while(!SerialUSB) {}                  // Wait until USB cable is connected
  delay(500);
 
  /*Serial1.begin(19200, SERIAL_8N1);              
  delay(5000);*/
	
	//Init Arm and set LED to on if error
  SerialUSB.print("Serial1->myARM.init: ");
	SerialUSB.println(myArm.Init(&Serial1));

	//Set Sigfox mode in uplink.
	SerialUSB.print("Set Mode:");
  SerialUSB.println(myArm.SetMode(ARM_MODE_SFX));
  SerialUSB.print("Downlink Mode Enabled: ");
	SerialUSB.println(myArm.SfxEnableDownlink(true));
  SerialUSB.print("Update Config Mode: ");
  SerialUSB.println(myArm.UpdateConfig());
	
  delay(5000); 
} // End of SETUP()

void loop()   {
	unsigned int i;

  // Print SIGFOX config
//  int id=0;
//  myArm.Info(0, 0, &id, 0, 0);
//  SerialUSB.print("ID= ");
//  SerialUSB.println(id, HEX);

	//Send the message to sigfox
	SerialUSB.print("Sigfox Send result: ");
	SerialUSB.println(myArm.Send(&msg, sizeof(msg)-1));

  //Wait (60s of timeout) and receive the data from sigfox
  myArm.Receive(data, sizeof(data), 60000);
  
	//Wait 10min
	for(i=0; i<10; i++) {
		delay(60000);
    SerialUSB.println(i);
	}
  SerialUSB.println("\n");
} // End of LOOP()
```

And this is the result answers through the USB port:
```
Serial1->myARM.init: 0
Set Mode:0
Downlink Mode Enabled: 0
Update Config Mode: 0
Sigfox Send result: 12
0
1
2
3
4
5
6
7
8
9

Sigfox Send result: 12
0
```
In the backend I see the follwoing messages:

1. The first message is empty, I receive 1 byte =00, the following messages are correct: 48656c6c6f20536967666f78

2. The messages are always uplink. Noone is sent in downlink mode.

Can you please help me?

antoine163, 2016-10-24:
What is your firmware version ?
What is version of armapi you used ?

joaqrus, 2016-10-24:
**Firmware version:** 
when I connect the N8 module through the serial port (19200,8,N,1), I get the following version number:

`ATV`

`REV. 590A S/N: 000F6F66 SFX 868MHZ 14DBM`

firmware was updated by me manually two weeks ago, following instructions received by mail.

**ARMAPI version:**

version=1.0.4
I can see inside the arm.cpp code that the files were generated last **Sep 26 2016** 
I downloaded the library from the github.io web page: [http://atim-radiocommunications.github.io/armapi/](url) Arduino Lib buttom.

antoine163, 2016-10-27:
There is a bug in '590A' and the armapi can't work with this firmware.
There are 3 way:
- Wait a new firmware.
- Use directly AT command 'AT$SF'.
- Get armapi from develop branch to use _armSfxSendReceive_. In this case you must make Arduino package.

joaqrus, 2016-10-28:
OK, thanks for the information. Can you please tell when you are expecting a new firmware release approx.?, I would like to adjust my project timeline.

Also I  would be interested in making the Arduino package. Can you please give me some clues about how to do it?

antoine163, 2017-05-09:
You can found a no official package from :
[http://atim-radiocommunications.github.io/armapi/pkg/Special/arduino_armapi_dev_v1.0.4bis.zip](http://atim-radiocommunications.github.io/armapi/pkg/Special/arduino_armapi_dev_v1.0.4bis.zip)
And use SfxSendReceive() method, please get the documentation about this method from arm.h file.

closed 2017-05-16 by antoine163