atim-radiocommunications/armapi #6: ARM_ERR_ARM_GO_AT in arminfo (RPi 3 w/Jessie)

mroi-snoc, 2017-05-02:
Hi there,

The example arminfo init fails due to an ARM_ERR_ARM_GO_AT error.

Serial port configured to /dev/ttyAMA0 (this is the only one available) and reported as necessary to make.

Do you have some hints ?

Thanks a lot.

antoine163, 2017-05-02:
Hi,
On RPi 3 you will need to use the "/dev/ttyS0".
You can use the TTY variable in the Makefile: `make TTY=/dev/ttyS0`
Note: Don't forget to disable the RESET pin (gpio4).

mroi-snoc, 2017-05-03:
Hi,
Thanks for answering quick.
Unfortunately"/dev/ttyS0" on RPi3 with recent Jessie releases does not seem to lead to the chip ; here is the output of the example built with this port :
ERROR - armPortConfig: Input/output error
ARM Init failed!
ARM_ERR_PORT_CONFIG: 'Port Error, at the port configuring.'

Can you explain how to disable the GPIO4 reset pin ?

Thanks a lot,

Maximilien.

mroi-snoc, 2017-05-03:
Hi Antoine,
I confirm building with /dev/ttyAMA0 works fine if setting GPIO 4 to 1.
This can be done with [wiringPi gpio utility](http://wiringpi.com/the-gpio-utility/) or the following commands (logged in as root) :
```bash
echo 4 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio4/direction
echo 1 > /sys/class/gpio/gpio4/value
```

Hope this helps,

Maximilien.