atim-radiocommunications/armapi #7: ARM_ERR_ARM_GO_AT on RPI 1

Michel4OpenOpcUa, 2017-05-09:
Hello,
in order to test the device i do the following 
1- I built arminfo
2- i disabled the gpio 4 with echo command as you explained in another issue
3- i run via SSH sudo ./arminfo  i get the following error.
ARM Init failed!
ARM_ERR_ARM_GO_AT: 'ARM command Error, can't switch to AT command.'

Can you help ?
regards
Michel

antoine163, 2017-05-16:
Hi,

It is possible if the radio module is busy by sending a message to Sigfox for example.
You can drive the reset pin (during 10ms) before to call ./arminfo.

Michel4OpenOpcUa, 2017-05-16:
The problem was fix by issuing several time the echo command.
We also changed the arminfo code to receive a 64bit representation of the serial num.
Your 32bit cast is not correct for registration. But now we have a more critical issue with initialization sequence. Il will open another issue for it.
Regards
Michel