atim-radiocommunications/armapi #8: ARM_ERR_ARM_GO_AT sur Arduino uno R3

snsu24, 2017-05-15:
Bonjour,

Les exemples fournis sur /armapi retourne l'erreur suivante : 
ARM_ERR_ARM_GO_AT: 'ARM commend Error, can't switch to AT commend.'

J'utilise un Arduino Uno et un module xbee Lora Atim. La communication avec le module xbee est réalisé via un port série virtuel généré par <SoftwareSerial.h>, pour conserver le moniteur sur le port série initial.
 
Avez-vous une solution pour débloquer cette situation ?

Je vous remercie par avance,
Cordialement,
Sébastien

antoine163, 2017-05-16:
Bonjour,

La liaison série utiliser par le shield correspond à l'objet Serial sur Arduino uno.
Vous ne prouvez pas utiliser de port série virtuel. L'erreur que vous avez et donc tout à fait normale.

jimibi, 2017-07-03:
Bonjour,

Comment changer de port si on utilise une autre carte Arduino.

antoine163, 2017-07-10:
Bonjour,

Je ne suis pas sûr de comprendre ce que vous voulez faire.
Le shield Arduino a été conçue pour fonctionner avec les pins 0 et 1 de l'Arduino (un 'HardwareSerial' dans le code Arduino).
Un port de l'Armapi a été développé pour fonctionner avec un 'HardwareSerial'.
Si vous voulez changer de port, il vous faudra l'implémenter vous-même en re-implémenter les méthodes du fichier 'armport_arduino.cpp' se trouvent dans le dossier d'installation des bibliothèques Arduino.

jimibi, 2017-07-10:
Bonjour,
Je travaille avec un ESP32 qui peut faire tourner du code Arduino et qui possède 3 port Série Hardware mais le premier est réservé pour le Flashage du programme et le debug, je voulais utiliser le port 2 ou 3 pour dialoguer avec un module ARM Nano N8-LW sur carte Xbee socket.
Depuis j'ai modifié l'API en C pour travailler avec le serial 3. Cela fonctionne correctement maintenant.
Merci quand même de votre réponse.