atim-radiocommunications/armapi #9: ARM_ERR_WOR_ENABLE when sending a msg with LoraAn

nwercto, 2017-05-15:


antoine163, 2017-05-16:
Hi,
The armSend don't return armError_t type but hte nuber of byte write.
Please read the documentation about this function: [http://atim-radiocommunications.github.io/armapi/doc/html/a00026.html#ga5013a2da2f2a1fe04e995129a1053393](http://atim-radiocommunications.github.io/armapi/doc/html/a00026.html#ga5013a2da2f2a1fe04e995129a1053393)

nwercto, 2017-05-16:
Hi!

Thank you for your reply and... My bad! :)
The fact is the message was never caught by our LoRa gateway. So I thought there was an error as the return code was not 0.

I have know to understand why the gateway don't receive the message whereas it receives message from other LoRa devices!