atim-radiocommunications/armapi #12: ARM_ERR_ARM_GET_REG sur Arduino UNO R3

olivierfavre, 2017-07-13:
Bonjour,

L'exemple fourni sigfox_uplink sur /armapi me renvoie une erreur d'initialisation LED13 à HIGH.
En affichant l'erreur renvoyée, j'ai un ARM_ERR_ARM_GET_REG.

J'utilise un Arduino Uno et un shield ACW_DUINO Sigfox (Pico).

Auriez vous une idée ?
Est ce que je peux cabler directement la shield pour envoyer les commandes AT ?

Merci d'avance.
Cordialement,
Olivier

olivierfavre, 2017-07-14:
Bonjour,
Début de résolution : J'ai inséré un delay(8000) avant le init. la led 13 reste éteinte et la led du module sigfox après les 8s s'éteint, clignote deux fois puis se rallume.
Par contre toujours rien côté backend sigfox où le device est enregistré...
Pouvez vous me confirmer que les "comportements" LED sont conformes a une séquence d'init avec succès?
Merci d'avance
Olivier

olivierfavre, 2017-07-16:
Re bonjour,
ci dessous copie du moniteur série :

`⸮+++ATV ATQ +++ATH000 ATH002 ATH003 ATH004 ATH008 ATH012 ATH013 ATH014 ATH015 ATH020 ATH025 ATH026 ATH027 ATH032 ATH043 ATH044 ATH061 ATH062 ATH063 ATH064 ATH065 ATH066 ATH090 ATH092 ATH093 ATH097 ATI +++ATH000=10 ATI +++ATH000=50 ATI Hello SigfoxHello SigfoxHello SigfoxHello SigfoxHello SigfoxHello Sigfox`
... pourtant aucune réception sur le backend...
Olivier

antoine163, 2017-08-01:
Bonjour,

L'armapi ne supporte pas les modules Pico.
Seule les modules ARM_N8_LP (+ version sigfox), ARM_N8_LD et ARM_N8_LW sont supportés.