atim-radiocommunications/armapi #14: Incomplete Documentation for LoraWan

FofeuLabs, 2017-11-15:
// French below

Hello,

I am trying to use your module to establish a connection using Lora. However, your documentation is very cryptic to me. How are you supposed to set all the keys ? Using armLwIds ?

Yours sincerely,
Fofeu

// French now

Bonjour,

J'essaye d'établir une connection entre votre module et une Gateway Lora. Cependant, j'ai du mal à comprendre votre documentation. Comment est-on sensé utiliser les clefs ? En utilisant armLwIds ?

Cordialement,
Fofeu

DiamoondWolf, 2020-02-04:
Aucune solution ?...

ClementD74, 2021-07-05:
@Fofeu I met the same issue as you with this function.
You need to use the key given to you when you have bought the product. Apparently, they can no longer be change with  armLwIds();

You can ask help on the support here :  https://www.atim.com/fr/support-technique/

The following code exemple here with a small change at l.86 myArm.LwEnableOtaa(false); changed it to true ;)
- https://github.com/atim-radiocommunications/armapi/blob/master/exemples/arduino/network_uplink/network_uplink.ino

It will allow to test Lorawan with TheThingNetwork for instance.